daq-tools/kotori #2: Issue on raspbian stretch

ggeorges39, 2019-03-01:
I have this issue run kotori 0.15.0 on raspbian strtech2019-03-01 19:51:32,132 [root                     ] INFO   : Root configuration file is /etc/kotori/kotori.ini
2019-03-01 19:51:32,135 [root                     ] INFO   : Requested configuration files: /opt/kotori/.kotori.ini, /etc/kotori/kotori.ini
2019-03-01 19:51:32,156 [root                     ] INFO   : Expanded configuration files:  /opt/kotori/.kotori.ini, /etc/kotori/kotori.ini, /etc/kotori/apps-enabled/hives.ini
2019-03-01 19:51:32,166 [root                     ] INFO   : Used configuration files:      /etc/kotori/kotori.ini, /etc/kotori/apps-enabled/hives.ini
2019-03-01T19:51:32+0100 [kotori                             ] INFO: Starting Kotori version 0.15.0
2019-03-01T19:51:32+0100 [kotori                             ] INFO: Using configuration file /etc/kotori/kotori.ini
2019-03-01T19:51:32+0100 [kotori.core                        ] INFO: Enabling applications ['hives']
2019-03-01T19:51:32+0100 [kotori.core                        ] INFO: Starting application "hives"
2019-03-01T19:51:32+0100 [kotori.core                        ] CRITICAL: Error loading entrypoint "kotori.daq.application.mqttkit:mqttkit_application"
2019-03-01T19:51:32+0100 [kotori.core                        ] CRITICAL: Application entrypoint "kotori.daq.application.mqttkit:mqttkit_application" for "hives" not loaded
2019-03-01T19:51:32+0100 [kotori.core                        ] INFO: Enabling vendors []

amotl, 2019-03-03:
Dear @ggeorges39,

thanks for writing in. While we have been able to improve the situation slightly by finding a workaround already [1], we are now offering a more recent Debian package for Raspbian Stretch [2]. It has been built inside a Docker container running the arm32v7 architecture (`balenalib/armv7hf-debian:stretch-build`).

We will be happy if someone could test this package on a RaspberryPi machine, while it should probably also work on a Beaglebone.

Please note it would make sense to follow the regular installation instructions [4] beforehand, as these will bring in recent packages for InfluxDB, Grafana and Mosquitto.

With kind regards,
Andreas.

[1] https://community.hiveeyes.org/t/issue-with-kotori-on-raspberrypi-arm/1584
[2] https://community.hiveeyes.org/t/issue-with-kotori-on-raspberrypi-arm/1584/18
[3] https://packages.hiveeyes.org/hiveeyes/foss/testing/kotori-standard_0.21.1-1_armhf.deb
[4] https://getkotori.org/docs/setup/debian-quickstart.html

amotl, 2019-03-04:
Dear @ggeorges39,

thanks for confirming [1] everything now works as expected with the new `kotori-standard` package for the `arm32v7` (`armhf`) architecture published to the package repository at [2]. The installation instructions [3] have been updated slightly and there are more details to read about the release at [4].

Thanks again for raising our awareness about this issue.

With kind regards,
Andreas.

[1] https://community.hiveeyes.org/t/issue-with-kotori-on-raspberrypi-arm/1584/20
[2] https://packages.hiveeyes.org/hiveeyes/foss/debian/
[3] https://getkotori.org/docs/setup/debian-quickstart.html
[4] https://community.hiveeyes.org/t/kotori-0-22-6-released/1600

closed 2019-03-04 by amotl