daq-tools/kotori #7: Kotori service fails to start

TheOneWhoKnocks96, 2019-05-31:
Hi! I installed the latest kotori package (0.22.7) for amd64 in my Ubuntu 18.04 LTS machine, but the kotori service fails to start. My machine has the archicteture x86_64.
If anyone could please help.

● kotori.service
   Loaded: not-found (Reason: No such file or directory)
   Active: failed (Result: exit-code) since Thu 2019-05-30 14:56:34 UTC; 19h ago
 Main PID: 15597 (code=exited, status=1/FAILURE)

May 30 14:56:34 igup-be systemd[1]: kotori.service: Main process exited, code=exited, status=1/FAILURE
May 30 14:56:34 igup-be systemd[1]: kotori.service: Failed with result 'exit-code'.
May 30 14:56:34 igup-be systemd[1]: kotori.service: Service hold-off time over, scheduling restart.
May 30 14:56:34 igup-be systemd[1]: kotori.service: Scheduled restart job, restart counter is at 5.
May 30 14:56:34 igup-be systemd[1]: Stopped Kotori data acquisition and graphing toolkit.
May 30 14:56:34 igup-be systemd[1]: kotori.service: Start request repeated too quickly.
May 30 14:56:34 igup-be systemd[1]: kotori.service: Failed with result 'exit-code'.
May 30 14:56:34 igup-be systemd[1]: Failed to start Kotori data acquisition and graphing toolkit.

The logs from kotori are:

from .compat import unicode
  File "/opt/kotori/lib/python2.7/site-packages/twisted/python/compat.py", line 611, in <module>
    import cookielib
  File "/usr/lib/python2.7/cookielib.py", line 32, in <module>
    import re, urlparse, copy, time, urllib
  File "/usr/lib/python2.7/copy.py", line 52, in <module>
    import weakref
  File "/usr/lib/python2.7/weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref

Regards

amotl, 2019-06-01:
Dear @TheOneWhoKnocks96,

thanks for letting us know about this issue. We will try to look into that.

With kind regards,
Andreas.

amotl, 2019-06-01:
Looks like the internet already knows about the very same issue you are observing. The most valuable references seem to be:
- https://github.com/MDAnalysis/mdanalysis/issues/1739
- https://askubuntu.com/questions/981663/python2-7-broken-by-weakref-import-error-please-help

amotl, 2019-06-01:
The issue can be reproduced minimally by invoking
```
/opt/kotori/bin/python -c "import weakref"
```

By following [1] to upgrade the existing virtual environment through
```
virtualenv --python /usr/bin/python2.7 /opt/kotori
```
the command above stops yielding an error and Kotori will start after invoking `systemctl restart kotori`.

[1] https://github.com/MDAnalysis/mdanalysis/issues/1739#issuecomment-436543303

---

Until we can resolve the issue on the packaging process, we humbly ask you to fix the existing virtualenv on your machine by applying this workaround.

TheOneWhoKnocks96, 2019-06-03:
I upgraded the existing virtual environment through 
   virtualenv --python /usr/bin/python2.7 /opt/kotori

But when i restart the kotori service with systemctl i can't find the kotori service:
   Failed to restart kotori.service: Unit kotori.service not found.

amotl, 2019-06-03:
Sorry to hear about that. When installing the package,
```
dpkg -i /tmp/kotori_0.22.7-1_amd64.deb
```
we receive the following output:
```
Selecting previously unselected package kotori.
(Reading database ... 386202 files and directories currently installed.)
Preparing to unpack /tmp/kotori_0.22.7-1_amd64.deb ...
useradd: user 'kotori' already exists
Unpacking kotori (0.22.7-1) ...
Setting up kotori (0.22.7-1) ...
Created symlink /etc/systemd/system/kotori.service → /usr/lib/systemd/system/kotori.service.
Created symlink /etc/systemd/system/multi-user.target.wants/kotori.service → /usr/lib/systemd/system/kotori.service.
```

After that, the services pretends to be installed correctly
```
systemctl list-unit-files | grep kotori
kotori.service                         enabled
```

And after fixing the dependency woes as outlined above, it will also actually start flawlessly.

Currently, I'm a bit out of ideas what might be going wrong on your machine regarding appropriate setup of the systemd unit file.

TheOneWhoKnocks96, 2019-06-03:
it Worked!! Thank you very much

amotl, 2019-06-03:
Hi @TheOneWhoKnocks96,

do you have an idea what actually made things work after it failed before? Was it just an arbitrary hiccup? Anyway, good to hear it is now working for you.

Happy telemetry and with kind regards,
Andreas.

TheOneWhoKnocks96, 2019-06-03:
The service of Kotori started and is active and running but i checked the logs of kotori.log and it gives this:
from .arrow import Arrow
	  File "/opt/kotori/local/lib/python2.7/site-packages/arrow/arrow.py", line 11, in <module>
	    from dateutil import tz as dateutil_tz
	  File "/opt/kotori/local/lib/python2.7/site-packages/dateutil/tz/__init__.py", line 2, in <module>
	    from .tz import *
	  File "/opt/kotori/local/lib/python2.7/site-packages/dateutil/tz/tz.py", line 19, in <module>
	    import six
	ImportError: No module named six
	
2019-06-03T11:30:06+0000 [kotori.core                        ] INFO    : Enabling vendors []

amotl, 2019-06-03:
This is another error you are experiencing right now? Bummer! We will try to continue to improve this situation for users on Ubuntu 18. In the meanwhile, we want to apologize for the inconvenience.

amotl, 2019-06-03:
I would have recommended to just install the `six` module through
```
/opt/kotori/bin/pip install six
```

But unfortunately, this also bails out
```
pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
```

amotl, 2019-06-03:
We've created a gist [1] with an accompanying automated non-invasive patch which will mitigate the problems learned through this conversation including the most recent one.

[1] https://gist.github.com/amotl/31eec18126c3f86908629671ea9ecc4d
[2]
```
bash <(curl -s https://gist.githubusercontent.com/amotl/31eec18126c3f86908629671ea9ecc4d/raw/5a70de72cfdce71f2f4925f52a0b50566bda96c2/kotori-virtualenv-fix.sh)
```

TheOneWhoKnocks96, 2019-06-03:
Yes it gives that error. What could i do? There's another solution?

TheOneWhoKnocks96, 2019-06-03:
I started the script a it mitigated the most recent problem but it raised another problem also:
(kotori.log)
module = __import__(self.module_name, fromlist=['__name__'], level=0)
	  File "/opt/kotori/local/lib/python2.7/site-packages/kotori/daq/application/mqttkit.py", line 6, in <module>
	    from kotori.daq.services.mig import MqttInfluxGrafanaService
	  File "/opt/kotori/local/lib/python2.7/site-packages/kotori/daq/services/mig.py", line 18, in <module>
	    from kotori.daq.storage.influx import InfluxDBAdapter
	  File "/opt/kotori/local/lib/python2.7/site-packages/kotori/daq/storage/influx.py", line 4, in <module>
	    import requests
	ImportError: No module named requests
	
2019-06-03T13:28:58+0000 [kotori.core                        ] INFO    : Enabling vendors []

What could be a better solution? Install kotori in a Debian strecth system?

amotl, 2019-06-03:
> What could be a better solution? Install kotori in a Debian stretch system? 

For now, that would the thing I would recommend to do, absolutely!

TheOneWhoKnocks96, 2019-06-03:
Okay, thank you very much for the help!!

TheOneWhoKnocks96, 2019-06-05:
Hi again!!
Sorry for bothering, but i installed kotori in a machine with Debian 9.9 (stretch) and it still gives the same problem:
(kotori.og)

from .arrow import Arrow
	  File "/opt/kotori/lib/python2.7/site-packages/arrow/arrow.py", line 11, in <module>
	    from dateutil import tz as dateutil_tz
	  File "/opt/kotori/lib/python2.7/site-packages/dateutil/tz/__init__.py", line 2, in <module>
	    from .tz import *
	  File "/opt/kotori/lib/python2.7/site-packages/dateutil/tz/tz.py", line 19, in <module>
	    import six
	ImportError: No module named six
	
2019-06-05T14:00:40+0100 [kotori.core                        ] INFO    : Enabling vendors []

TheOneWhoKnocks96, 2019-06-05:
i solved it for now!! I imported the arrow from PyPi

amotl, 2019-06-05:
Hi there,

bummer. We haven't exactly been aware that the Debian package actually causes problems on Debian!

I just checked and want to add that we are successfully running Kotori on a Debian 9.9 system where both the `arrow` and `six` packages seem to be part of the Python virtualenv as intended:
```
/opt/kotori/bin/python

>>> import six
>>> six.__file__
'/opt/kotori/lib/python2.7/site-packages/six.pyc'

>>> import arrow
>>> arrow.__file__
'/opt/kotori/lib/python2.7/site-packages/arrow/__init__.pyc'
```

Nevertheless, I am still curious where this issue might be coming from and definitively want to resolve this for other users.

In the meanwhile, we are happy that you have been able to help yourself. Do you remember what you did **exactly** to mitigate this installation issue on Debian? Maybe post-installing `arrow` and/or `six` from PyPI was successfully like
```
/opt/kotori/bin/pip install arrow six
```

With kind regards,
Andreas.

TheOneWhoKnocks96, 2019-06-07:
What i did was to install the arrow package:
$ pip install -U arrow

amotl, 2019-06-07:
Thanks for letting us know. Did you install it into the virtualenv or into the system Python?

amotl, 2020-01-10:
Dear @Ozymandias96,

Kotori 0.23.0 packages are now available for Debian buster. They might well also work on Ubuntu 18.04.

The package repository is at [1]:
```
deb https://packages.elmyra.de/elmyra/foss/debian/ buster main foundation
```

With kind regards,
Andreas.

[1] https://getkotori.org/docs/setup/debian-quickstart.html#register-with-package-repository

Dewieinns, 2020-02-29:
I just wanted to chime in for others who may be trying to make this work

On a fresh install of Debian 10.3 (I tried Ubuntu initially also - encountered above issues as well), following the [Quick Start] (https://getkotori.org/docs/setup/debian-quickstart.html) I still had the issue with the: 
`ImportError: cannot import name _remove_dead_weakref`
error. I was able to resolve it with the suggested:
`virtualenv --python /usr/bin/python2.7 /opt/kotori`

I then encountered the next issue:
`ImportError: No module named six`
and I attempted to run:
`/opt/kotori/bin/pip install arrow six`
but I encountered:
`root@debianDepthData:~# /opt/kotori/bin/pip install arrow six
/opt/kotori/share/python-wheels/requests-2.21.0-py2.py3-none-any.whl/requests/__init__.py:91: RequestsDependencyWarning: urllib3 (1.24.1) or chardet (2.3.0) doesn't match a supported version!
pip is configured with locations that require TLS/SSL, however the ssl module in Python is not available.
`

I've been searching for a solution to this error but everything seems to point to updating libraries for Python3 and it is my understanding you're using Python 2.7? I'm no programmer but know just enough to be dangerous. It appears I maybe don't have a full-blown version of pip?

amotl, 2020-02-29:
Hi @Dewieinns,

thanks for writing in.

1. Did you actually use
```
deb https://packages.elmyra.de/elmyra/foss/debian/ buster main foundation
```
as the package repository?

2. Can you tell me what is the `kotori*.deb` package name within `/var/cache/apt/archives` on your machine?

With kind regards,
Andreas.

Dewieinns, 2020-03-01:
Hey @amotl , 

I have started my [own issue](https://github.com/daq-tools/kotori/issues/22) so as not to flood this one with useless information however I can tell you that the file in the package is:

kotori_0.23.0-1~buster_amd64.deb

Dewieinns, 2020-03-01:
Time for another Fun filled Ubuntu update:

Today I started over fresh with my Ubuntu install and tried to document as many variations as I could. 

First off - [Register with Package Repository](https://getkotori.org/docs/setup/debian-quickstart.html#register-with-package-repository)

`apt-add-repository 'deb https://packages.elmyra.de/elmyra/foss/debian/ buster main foundation'`

Resulted in:
`The following signatures couldn't be verified because the public key is not available: NO_PUBKEY XXXXXXXXXXXXXXXXXXX`

so I discovered that if you add them **first** via:
`wget -qO - https://packages.elmyra.de/elmyra/foss/debian/pubkey.txt | apt-key add -`

you can then add the repo without errors

I then ran _apt update_ and  _apt upgrade_ to be sure everything was up to date before starting. 

When that was done I executed _apt install --install-recommends --install-suggests kotori_

During Install I had no idea what to enter for a few of these. **Could you please provide a bit of feedback here?**

- **Chief Server**
localhost:4000_

- **icecast2 & configuring servers**
Selected _No_

- **configuring jackd2** - mentioned lockups 
Selected _No_

- **Kerberos Authentication** _Often, the default realm is the uppercase version of the local DNS domain._ 
_MYDOMAIN.CA_

- **Kerberos Authentication servers in realm separated by spaces**
Entered nothing

- **Kerberos Administrative server in realm**
Entered nothing

- **Postfix Configuration**
Selected "_Internet Site_"

Everything seemed to be ok until I got to:
`tail -F /var/log/kotori/kotori.log`

`/opt/kotori/bin/python: /lib/x86_64-linux-gnu/libc.so.6: version 'GLIBC_2.28' not found (required by /opt/kotori/bin/python)`

so I went back and tried to upgrade the virtual environment	
`virtualenv --python /usr/bin/python2.7 /opt/kotori`

This required me to install _virtualenv_.

I did a bit more messing around and was still unable to get the Kotori service to start. I ended up issuing the above command again and finally, the service is up and running it says. 

I hop on over and have a look at the Kotori logs again and see the following in there:
```
        Source of exception:
        Filename:    /opt/kotori/local/lib/python2.7/site-packages/arrow/arrow.py
        Line number: 11
        Function:    <module>
        Code:        from dateutil import tz as dateutil_tz

        Traceback (most recent call last):
          File "/opt/kotori/local/lib/python2.7/site-packages/kotori/core.py", line 80, in get_application_factory
            factory_callable = self.load_entrypoint(app_factory)
          File "/opt/kotori/local/lib/python2.7/site-packages/kotori/core.py", line 102, in load_entrypoint
            thing = entrypoint.load(require=False)
          File "/opt/kotori/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2324, in load
            return self.resolve()
[... 55 lines cut ...]
   Active: inactive (dead)
     Docs: http://docs.grafana.org
```

so I start it and it seems to start no problem. I restart Kotori again and look in the logs:
```
2020-03-01T22:04:06+0000 [kotori.daq.graphing.grafana.api    ] WARN    : Problem getting folder uid="instagraf"": Unauthorized
2020-03-01T22:04:06+0000 [kotori.daq.graphing.grafana.api    ] WARN    : Problem creating instant folder: Unauthorized
```

Which isn't good. As per the Guide I had logged in to Grafana as admin/admin. I was prompted to set a password; maybe that's not good? I reset it back to "admin" to test my theory:

https://community.grafana.com/t/how-do-i-reset-admin-password/23

Restarted Kotori again - no warnings! no errors! So I again send a telemetry packet - again no errors! 

I then attempted to look at the dashboard and [Watch Telemetry Data](https://getkotori.org/docs/setup/getting-started.html#watch-telemetry-data) as per the Guide 

but the provided sample URL did not work for me (_http://192.168.1.132:3000/dashboard/db/ecuador_)

I was able to find it by _logging in to Grafana_ > _Dashboards_ > _Manage_ > _Instant Dashboards_ > _ecuador_

Success! Data! I am guessing this is maybe due to a newer version of Grafana? 

Moving forward I guess I have to determine how I can secure Grafana better (better password) and have Kotori still be able to access it.

amotl, 2020-03-01:
Dear @Dewieinns,

thanks for sharing your walkthrough and for your patience getting Kotori up and running. It's a bit poor that it caused so much hassle for you.

I should really update the documentation accordingly to make the installation work more smoothly. Apart from that, I should investigate a) why Kotori pulls in such a wrath of dependencies through the Debian package installation path which might not be required at all by improving the documentation to install specific packages explicitly and b) why it fails on installing the **essential** packages correctly within the Python virtualenv.

Apart from that, finally moving forward to Python 3 (#20) would do no harm either and might also improve the installation experience.

With kind regards,
Andreas.

Dewieinns, 2020-03-01:
Not a problem, I provided the feedback to hopefully help others moving forward (or perhaps you to improve documentation) If there is anything I can do to help please feel free to ask. I have a feeling I will have some more questions as I go :)

Dewieinns, 2020-03-01:
Yes a move to Python 3 I think would be a good idea. If I can get my
project up and going it will be initially small scale and I am quite
willing to help diagnose/test if you require assistance.

I am "Software Quality Assurance Automation Engineer" <-- (designated
title) by day, I'm pretty good at breaking things and then reporting how I
broke them...

On a side note - a friend of mine has bees and we've been talking about
developing a solution similar to what you folks seem to have developed. I
ran across your software pursuing an imitative of my own knowing it would
help produce something for him; finding something you've made which will
help me help him is even better. I knew the temperature/humidity/etc
would be fairly straight forward but it was great to see solutions being
shared about the load sensors.

I'm curious how "many" of you are involved as "The Kotori Developers" and
from where you originate?

- Andrew

amotl, 2020-03-01:
Dear @Dewieinns,

I've slightly updated the documentation [1] through dfa62db1 and the gist on [2]. Thanks for your feedback!

With kind regards,
Andreas.

[1] https://getkotori.org/docs/setup/debian-quickstart.html
[2] https://gist.github.com/amotl/31eec18126c3f86908629671ea9ecc4d

amotl, 2020-03-01:
> On a side note - a friend of mine has bees and we've been talking about
developing a solution similar to what you folks seem to have developed. I
ran across your software pursuing an imitative of my own knowing it would
help produce something for him; finding something you've made which will
help me help him is even better.

You are welcome to join us on https://community.hiveeyes.org/.

> I'm curious how "many" of you are involved as "The Kotori Developers" and
from where you originate?

We are mostly from Berlin, Germany. You are from Canada? Cheers!

Dewieinns, 2020-03-01:
You are welcome to join us on https://community.hiveeyes.org/.

We will!

> We are mostly from Berlin, Germany. You are from Canada? Cheers!
Indeed we are. I see now I'm replying to the github issue thread and will
stop. Thanks again!

Dewieinns, 2020-03-02:
I just installed this again using the minimum set of required packages. Even after executing your script I still ended up having to run:

`/opt/kotori/bin/pip install arrow six`

to get dateutil to download for some reason.. 

`sudo /opt/kotori/bin/pip install arrow six
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. A future version of pip will drop support for Python 2.7. More details about Python 2 support in pip, can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support
WARNING: The directory '/home/dewie/.cache/pip' or its parent directory is not owned or is not writable by the current user. The cache has been disabled. Check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Requirement already satisfied: arrow in /opt/kotori/lib/python2.7/site-packages (0.10.0)
Requirement already satisfied: six in /opt/kotori/lib/python2.7/site-packages (1.12.0)
Collecting python-dateutil
  Downloading python_dateutil-2.8.1-py2.py3-none-any.whl (227 kB)
     |████████████████████████████████| 227 kB 572 kB/s
ERROR: kotori 0.23.0 requires pyparsing==2.2.0, which is not installed.
ERROR: bokeh 0.12.4 requires numpy>=1.7.1, which is not installed.
ERROR: pandas 0.18.1 requires numpy>=1.7.0, which is not installed.
ERROR: matplotlib 2.0.0 requires cycler>=0.10, which is not installed.
ERROR: matplotlib 2.0.0 requires numpy>=1.7.1, which is not installed.
ERROR: matplotlib 2.0.0 requires pyparsing!=2.0.0,!=2.0.4,!=2.1.2,!=2.1.6,>=1.5.6, which is not installed.
ERROR: matplotlib 2.0.0 requires subprocess32, which is not installed.
ERROR: ggplot 0.9.7 requires cycler, which is not installed.
ERROR: ggplot 0.9.7 requires numpy, which is not installed.
ERROR: ggplot 0.9.7 requires scipy, which is not installed.
Installing collected packages: python-dateutil
Successfully installed python-dateutil-2.8.1
`

amotl, 2020-03-02:
Dear @Dewieinns,

thanks for your feedback. This is really weird. If you will have another chance, I will be happy to see the output of the fixup script.

Cheers,
Andreas.

amotl, 2020-03-03:
While we haven't been able to build appropriate packages for Ubuntu, Kotori 0.24.5 is now available as a Docker image from Docker Hub, see also [1] and [2].

However, Debian "stretch" 9 and Debian "buster" 10 are supported natively by corresponding Debian packages on the package repository as outlined on [3].

[1] https://hub.docker.com/repository/docker/daqzilla/kotori
[2] https://getkotori.org/docs/setup/docker.html
[3] https://getkotori.org/docs/setup/debian-quickstart.html

amotl, 2020-03-04:
Dear @RuiPinto96 and @Dewieinns,

we finally added builds for Ubuntu 18 Bionic Beaver and updated the documentation at [1] accordingly. We have been able to give it a quick test on a VirtualBox machine but will also be happy to receive feedback from you.

With kind regards,
Andreas.

[1] https://getkotori.org/docs/setup/debian-quickstart.html#register-with-package-repository

closed 2020-03-06 by amotl