daq-tools/kotori #8: Add adapter for ingesting data from TTN

amotl, 2019-06-03:
@thiasB asked the other day
> Did I mention that TTN can also send the data to an HTTP endpoint [1]? It would be nice if Kotori would accept that, too.

[1] https://www.thethingsnetwork.org/docs/applications/http/

amotl, 2019-06-03:
We are tracking this at
- https://community.hiveeyes.org/t/ttn-daten-an-kotori-weiterleiten/1422

and others also asked about related things:
- https://www.thethingsnetwork.org/forum/t/labs-store-and-visualize-data-using-influxdb-and-grafana/5297/22

amotl, 2020-03-03:
@thiasB added the appropriate implementation to https://github.com/daq-tools/ttnlogger the other day. Thanks!

At some point, we will port that into an appropriate decoder implementation for Kotori.

amotl, 2023-05-24:
Hi again,

we made some progress by finally integrating GH-81. GH-132 is still a bit dormant, but GH-133 has further ideas on this topic, in order to generalize a corresponding implementation.

With kind regards,
Andreas.

amotl, 2023-05-24:
@ClemensGruber is working on [^1] and just shared the TTN payload of a [Paxcounter] device with us at [^2]. Thank you.

We will have to check what happens with that data payload fragment, which contains a list of `bytes`, which are not quite typical telemetry/metric/measurement values.

```json
"decoded_payload": {
  "ble": 1,
  "bytes": [
    1,
    0,
    1,
    0
  ],
  "pax": 2,
  "port": 1,
  "wifi": 1
}
```

[^1]: https://community.hiveeyes.org/t/pax-counter-mit-lopy4/4838
[^2]: https://community.hiveeyes.org/t/tts-ttn-daten-an-kotori-weiterleiten/1422/87

[Paxcounter]: https://github.com/cyberman54/ESP32-Paxcounter

ClemensGruber, 2023-05-24:
> We will have to check what happens with that data payload fragment,

No need to use `bytes `array, I can avoid them via the TTN payload decoder, see https://community.hiveeyes.org/t/tts-ttn-daten-an-kotori-weiterleiten/1422/90 n

amotl, 2023-05-24:
> No need to use `bytes` array.

Yes, my note is about properly _ignoring_ it.