daq-tools/kotori #11: Weather Variables (Davis Vantage Pro2) Weewx

TheOneWhoKnocks96, 2019-06-26:
Hello!! I am utilizing a Davis Vantage Pro2 station with Weewx, Kotori, Mosquitto, InfluxDB and Grafana architecture to save and display the weather data. 
I don't know if anyone can help me... My issue is not really an issue related with kotori and the whole architecture. That is working fine. My issue is about understanding what some values of the Davis Vantage Pro 2 variables mean. Some variables it's obvious what they meant but some i have no clue (for example: rxCheckPercent, Sunset, ForecastRule, ForecastIcon, stormStart, etc). 
I looked in the Davis Vantage manuals but it not tells that much. I know the issue it's not directly related with Kotori, but i thought you guys might help.

Here's the JSON example that you have in the kotori website of all variables that the console gives:

    "windSpeed10_kph": "5.78725803977",
    "monthET": "1.32",
    "highUV": "0.0",
    "cloudbase_meter": "773.082217509",
    "leafTemp1_C": "8.33333333333",
    "rainAlarm": "0.0",
    "pressure_mbar": "948.046280104",
    "rain_cm": "0.0",
    "highRadiation": "0.0",
    "interval_minute": "5.0",
    "barometer_mbar": "1018.35464712",
    "yearRain_cm": "17.2000000043",
    "consBatteryVoltage_volt": "4.72",
    "dewpoint_C": "2.07088485785",
    "insideAlarm": "0.0",
    "inHumidity": "29.0",
    "soilLeafAlarm4": "0.0",
    "sunrise": "1492489200.0",
    "windGust_kph": "9.65608800006",
    "heatindex_C": "3.55555555556",
    "dayRain_cm": "0.0",
    "lowOutTemp": "38.3",
    "outsideAlarm1": "0.0",
    "forecastIcon": "8.0",
    "outsideAlarm2": "0.0",
    "windSpeed_kph": "3.95409343049",
    "forecastRule": "40.0",
    "windrun_km": "1.07449640224",
    "outHumidity": "90.0",
    "stormStart": "1492207200.0",
    "inDewpoint": "45.1231125123",
    "altimeter_mbar": "1016.62778614",
    "windchill_C": "3.55555555556",
    "appTemp_C": "1.26842313302",
    "outTemp_C": "3.55555555556",
    "windGustDir": "275.0",
    "extraAlarm1": "0.0",
    "extraAlarm2": "0.0",
    "extraAlarm3": "0.0",
    "extraAlarm4": "0.0",
    "extraAlarm5": "0.0",
    "extraAlarm6": "0.0",
    "extraAlarm7": "0.0",
    "extraAlarm8": "0.0",
    "humidex_C": "3.55555555556",
    "rain24_cm": "0.88000000022",
    "rxCheckPercent": "87.9791666667",
    "hourRain_cm": "0.0",
    "inTemp_C": "26.8333333333",
    "watertemp": "8.33333333333",
    "trendIcon": "59.7350993377",
    "soilLeafAlarm2": "0.0",
    "soilLeafAlarm3": "0.0",
    "usUnits": "16.0",
    "soilLeafAlarm1": "0.0",
    "leafWet4": "0.0",
    "txBatteryStatus": "0.0",
    "yearET": "4.88",
    "monthRain_cm": "2.94000000074",
    "UV": "0.0",
    "rainRate_cm_per_hour": "0.0",
    "dayET": "0.0",
    "dateTime": "1492467300.0",
    "windDir": "283.55437192",
    "stormRain_cm": "1.72000000043",
    "ET_cm": "0.0",
    "sunset": "1492538940.0",
    "highOutTemp": "38.4",
    "radiation_Wpm2": "0.0"

amotl, 2019-06-26:
Dear @TheOneWhoKnocks96,

hello again and thanks for writing in.

Your question resonates very much with us. While this is not on the list of our top priorities, we would really like to work on a decent visualization of weather information coming from devices like your are using with WeeWX.

@wetterfrosch might be interested in general about this topic and usually knows how to decipher arbitrary database and telemetry fields from the area of weather information but I can't promise anything about his interest to spend time on this and on his timeframe itself.

Saying that, I would like to thank you very much for opening this as an issue as I always wanted to have a description table at hand which tells details about what is actually put into these fields.

> I looked in the Davis Vantage manuals but it not tells that much.

Maybe it's more related to WeeWX and you should scan their documentation? Honestly, I don't know that much details yet, but @tonke might also be able to help here.

With kind regards,
Andreas.

TheOneWhoKnocks96, 2019-06-26:
Thanks a lot for helping!!

Volgi84, 2019-08-24:
Hello TheOneWhoKnocks96,
you might be more successful searching in the weewx forums. 
[Here](http://www.weewx.com/docs/customizing.htm#Translate_labels_and_modify_local_options) is a link where there are some values shown in clear text. Just search that page and it might give some insight. 
Best Regards, 
Volker

amotl, 2019-08-27:
Dear @Volgi84,

thanks for pointing out the relevant resources to us, @wtf might also be interested in these details. Also thanks for mentioning the WeeWX forums [1,2] as we haven't been aware of them.

With kind regards,
Andreas.

[1] https://groups.google.com/forum/#!forum/weewx-user
[2] https://groups.google.com/forum/#!forum/weewx-development