daq-tools/kotori #12: kotori question (doubt)

TheOneWhoKnocks96, 2019-07-01:
Hello again!! Sorry to bother... I have a question about how Kotori exports data to influxDB.
I'm writing a thesis about the system of my automatic acquisition of meteorological data from the Davis Vantage Pro2 station and i'm using your system.
I would like to know more about Kotori exports data to influxDB. Is it through http? Through GET request? What kotori does to the JSON that comes from MQTT?

I kinda know a little of how it works but i wanna have certainty.
Best Regards.

amotl, 2019-07-01:
Dear @TheOneWhoKnocks96,

thanks for asking, I will be happy to answer your question appropriately. In turn, may I ask you more about the background of the thesis you are writing? I am also interested in that, so we could mutually share our knowledge given that you are allowed to talk about any details yet.

With kind regards,
Andreas.

amotl, 2019-07-02:
> I would like to know more about Kotori exports data to influxDB. Is it through http? Through GET request? What kotori does to the JSON that comes from MQTT?

While this is far away from a thorough answer I would like to give you regarding your question, these posts give some insights about what is going on under the hood at least.

https://community.hiveeyes.org/t/how-do-measurements-flow-from-mqtt-to-grafana/729

amotl, 2020-03-10:
Dear @RuiPinto96,

thank you very much for the conversation we had about the internals of Kotori the other day. I finally have been able to refactor the contents into a FAQ section within the documentation, see [1] and [2]. If you feel this doesn't answer your questions yet, don't hesitate to reopen.

With kind regards,
Andreas.

[1] https://getkotori.org/docs/faq/acquisition.html
[2] https://getkotori.org/docs/faq/weewx.html

closed 2020-03-10 by amotl