daq-tools/kotori #15: Mosquitto authentication

TheOneWhoKnocks96, 2019-08-29:
Hello!!
I'm trying to connect with Kotori to Mosquitto through authentication and i'm not able to connect.
It gives me: Error connecting to MQTT broker but retrying each 5 seconds

I enabled the allow_anonymous to false and defined a user and password in a file in order to connect to mosquitto.
Then, i changed the user and password in the Kotori.ini file and restarted Kotori but it will not connect to the mqtt broker. 

; mqtt bus adapter
[mqtt]
host        = localhost
#port        = 1883
username    = rui
password    = ********

Best Regards

amotl, 2019-08-29:
Dear @TheOneWhoKnocks96,

thanks for reporting your observations. We will have to review the situation about authenticating Kotori with the MQTT broker.

Please feel free to ping us again if you don't hear back from us soon.

With kind regards,
Andreas.

TheOneWhoKnocks96, 2019-08-29:
It gives me this log:

2019-08-29T11:53:30+0100 [kotori.daq.intercom.mqtt.paho ] CRITICAL: Error connecting to MQTT broker but retrying each 5 seconds

It should work because the authentication is done here in https://github.com/daq-tools/kotori/blob/master/kotori/daq/intercom/mqtt/paho.py#L37:
        # Optionally authenticate connection
         if self.broker_username: 
              self.client.username_pw_set(self.broker_username, self.broker_password)

Best Regards and thanks for responding.

Ozymandias96, 2019-09-02:
Dear @amotl 

Have you resolved the situation yet?

With kind regards.

Ozymandias96, 2019-09-04:
Is it in here https://github.com/daq-tools/kotori/blob/master/kotori/daq/intercom/mqtt/base.py?

@amotl please help

Ozymandias96, 2019-09-04:
It works fine, i changed to a password with numbers and it started to work...

amotl, 2019-09-04:
Dear @Ozymandias96,

> Have you resolved the situation yet?

Sorry, I am currently traveling.

> It works fine, i changed to a password without numbers and it started to work...

Seriously? I am glad this helped for you but you made me curious about this situation: Can you verify it really does not work when using passwords containing numbers?

Thanks already and with kind regards,
Andreas.

amotl, 2019-11-07:
Dear @Ozymandias96,

may I humbly ask you again whether using passwords containing numbers makes things go south in any way?

I will happy to investigate, but want to get more feedback on this first.

With kind regards,
Andreas.

amotl, 2020-03-06:
Dear @RuiPinto96,

may I ask you again about the behavior you have been observing here the other day? Did your password just **contain** numbers or was it solely made of **numbers only**?

With kind regards,
Andreas.

amotl, 2021-03-02:
Dear Rui,

long time no see, I hope you are doing well.

I just wanted to express my happiness to see Kotori mentioned so extensively within your MSc paper "Automatic Data Acquisition from the Meteorological Station at IGUP". I really liked to read about the [WeeWX](https://getkotori.org/docs/gallery/weewx.html)-scenario being described and elaborated in such a thorough manner.

With kind regards,
Andreas.

---

- https://hdl.handle.net/10216/127880
- https://repositorio-aberto.up.pt/handle/10216/127880
- https://repositorio-aberto.up.pt/bitstream/10216/127880/2/409346.pdf

RuiPinto96, 2021-03-07:
Hi Andreas,

Fortunately, I am doing well and hope your doing well too.

Thank you, I didn't have the opportunity but I want to thank you for the help. Kotori is a great data acquisition tool and it really helped me in my MSc project.

Best regards and stay safe
Rui.