daq-tools/kotori #17: Running Kotori within Docker

amotl, 2019-11-29:
While we use Docker already for **building** Debian packages for Kotori [1], we just learned from #16 that @agross is also using it for actually **running** Kotori.

Thinking about that, it would actually be cool to have a `docker-compose.yml` file ready for running a complete environment of Mosquitto, InfluxDB, Grafana and Kotori on your fingertips, right?

Community contributions regarding this are always welcome. Maybe you will be able to volunteer on that, @agross? ;]

[1] https://www.getkotori.org/docs/setup/debian-quickstart.html

agross, 2019-11-29:
https://github.com/agross/weewx-kotori-influxdb-grafana

amotl, 2019-11-29:
That was quick. Thanks a bunch!

agross, 2019-11-29:
You're welcome! It would be great to support Grafana API key authentication. Seems like the Grafana lib supports it, but Kotori requires user/pass. Probably worth another issue 😉