daq-tools/kotori #22: Installation/Configuration on fresh install of Debian 10 fails.

Dewieinns, 2020-03-01:
Hey @amotl  I am starting a new thread to follow up with a post I made where [similar was happening](https://github.com/daq-tools/kotori/issues/7) as I'm experiencing slightly different variations now. 

After my reply last evening I set about working with a new VM. I didn't properly document everything I had done so this morning I started over **COMPLETELY** fresh. I was seeing weird issues where systemctl didn't seem to be running and the VM I was using had only one cpu/core. I assumed this was the reason for performance issues I was seeing so I wiped it, made a new VM (on a different drive in my server even) and set about installing Debian. 

I opted _not to install the GUI_ and _enabled the SSH server_. 

With Debian successfully installed I installed _screen_ and then started following the [Setup on Debian](https://getkotori.org/docs/setup/debian-quickstart.html) guide again. 

**Note:** When I started following this guide initially (yesterday) it wasn't obvious to me (I'm a n00b) that I needed to install the package source for Debian Stretch (9) _OR_ Debian Buster (10). 

This time I added only the package source I needed (Buster) and was good to go... until:

_Add GPG key for checking package signatures:
wget -qO - https://packages.elmyra.de/elmyra/foss/debian/pubkey.txt | apt-key add -_

Error:
`E: gnupg, gnupg2 and gnupg1 do not seem to be installed, but one of them is required for this operation`

Fix:
`apt-get install gnupg`

All is good and I set about installing Kotori as well as recommended and suggested packages (14.9GB - takes about an hour and a half on my slow internet connection)

Next thing prompted for is "_Configuring jackd2_"

_Do I want to run with realtime priority? (_explains it may lead to lock-ups.) <-- Selected No

then followed:

**_Configuring Kerberos Authentication_**
_Default Kerbors version 5 realm:_
	- Pre-populated with my domain <-- left it as is
 _Kerberos servers for your realm:_
	- Nothing pre-populated <-- left it blank
 _Administrative server for your Kerberos realm:_
	- Nothing pre-populated <-- left it blank

It then goes through and installs everything - this takes some time - until it gets to the end where there are a couple of errors displayed:

```
Setting up mh-e (8.5-2.1) ...
ERROR: mh-e is broken - called emacs-package-install as a new-style add-on, but has no compat file.
```

and then:

```
Errors were encountered while processing:
 lirc
 lirc-x
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

At this point I rebooted the VM using _systemclt reboot_

After reboot I noticed VM was SLOW again - mega slow. 

I again ran _apt update_, all packages were up to date. 

I then again ran _apt install --install-recommends --install-suggests kotori_ and noticed the following:

```
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
2 not fully installed or removed.
```

I hit Y to continue and it looked like everything was successful this time. 

I attempted to execute _systemctl start influxdb grafana-server_ but received the following warning again:

```
System has not been booted with systemd as init system (PID 1). Can't operate.
Failed to connect to bus: Host is down
```

It was after this that I wiped the VM last time thinking I had done something mega wrong... 

I then installed _open-vm-tools_ on the VM which allowed me to gracefully reboot the system via the VM Host console. 

Upon rebooting systemctl still isn't running... at this point I'm kind of at a loss. I was farther than this trying to install it on Ubuntu yesterday.  

Sorry for the wall of information but I wanted to be thorough enough you may be able to provide some insight as to what's going on (Hopefully something more than "that's really weird" haha)

If you wish me to get in touch directly I can do this also.

amotl, 2020-03-01:
Dear @Dewieinns,

thanks for your persistence!

I recognize the full list of suggested dependencies of Kotori is pretty huge. However, you should be able to run it with a minimal set of dependencies without using `--install-recommends --install-suggests`. Then, you should install InfluxDB and Grafana manually, probably by just invoking
```
apt-get install influxdb grafana
```

Thanks for figuring out the requirement of GnuPG. I can't tell anything about any VM weirdness you might be observing.

> If you wish me to get in touch directly I can do this also.

Sure, just drop me an email at andreas@getkotori.org.

With kind regards,
Andreas.

Dewieinns, 2020-03-01:
Just wanted to give you a heads-up that I've sent 2 email messages now; just in the even your SPAM filter nabbed them.

amotl, 2020-03-03:
Dear @Dewieinns,

we updated the packages for Debian "buster" 10, so Kotori should now install without pulling in such a large number of dependencies you have been observing at https://github.com/daq-tools/kotori/issues/7#issuecomment-593154679.

With kind regards,
Andreas.

[1] https://getkotori.org/docs/setup/debian-quickstart.html

closed 2020-03-04 by amotl