daq-tools/kotori #28: Fix "Connection pool is full"

adrianlzt, 2020-03-21:
A simple configuration option while creating the InfluxDB client should be enough.
https://influxdb-python.readthedocs.io/en/latest/api-documentation.html#influxdbclient

```
pool_size (int) – urllib3 connection pool size, defaults to 10.
```
Added [here](https://github.com/daq-tools/kotori/blob/master/kotori/daq/storage/influx.py#L47)

amotl, 2020-03-21:
Dear Adrián,

thanks for writing in and for using Kotori. May I ask you how you encountered this problem?

With kind regards,
Andreas.

adrianlzt, 2020-03-21:
Just a pet project of an arduino sending data each 12.5ms

amotl, 2020-03-21:
Thanks for the details. So, you are just sending a high throughput of datapoints to the MQTT bus which quickly triggers this condition and you are observing this through log messages on STDERR?

I will see what I can do against that. Do you believe that raising the pool size will solve the problem? Have you been able to solve it that way? And if so, which pool size did you use?

I am just thinking about what will happen if the pool is exhausted again. Shall we think about introducing queues?

adrianlzt, 2020-03-21:
Yes, I increased the pool to 50 and it worked without warnings.
I think just giving the option to users could be enough.

amotl, 2020-03-21:
All right, thanks! I will add `pool_size` as an option to the InfluxDB configuration then.

Nevertheless, I also would like to point out that you can configure Kotori to use the UDP protocol to talk to InfluxDB (see [1] and the `use_udp` option within [2]) in order to improve the acquisition throughput by getting rid of the overhead coming from HTTP.

However, this feature is not obvious, poorly documented and a bit weird to configure, see [3-5]. I will see what I can do to improve this if you would like to use that feature to increase your data rate without wasting resources.

[1] https://docs.influxdata.com/influxdb/v1.7/supported_protocols/udp/
[2] https://influxdb-python.readthedocs.io/en/latest/api-documentation.html#influxdbclient
[3] https://getkotori.org/docs/gallery/luftdaten.info/index.html#id2
[4] https://github.com/daq-tools/kotori/blob/master/kotori/daq/storage/influx.py#L28-L29
[5] https://github.com/daq-tools/kotori/blob/master/kotori/daq/storage/influx.py#L39-L43

adrianlzt, 2020-03-21:
Yes, I saw that option, but, correct me if I'm wrong, but only second
precision is available and I need ms

amotl, 2020-04-12:
Dear Adrián,

> but only second precision is available and I need to express milliseconds

You might be correct on that important detail and now I remember it when using it the other day. Regarding this issue, I found https://github.com/influxdata/influxdb/issues/841 and https://github.com/influxdata/influxdb/issues/8746 on the InfluxDB issue tracker.

The latter one refers to a `precision` options which could be used inside the `[[udp]]` section of the configuration file like
```
[[udp]]
  # ...

  # InfluxDB precision for timestamps on received points ("" or "n", "u", "ms", "s", "m", "h")
  # precision = ""
```

-- https://github.com/influxdata/influxdb/issues/8746#issuecomment-325507363

However, there are still many (unresolved) issues revolving around this topic:
- https://github.com/influxdata/influxdb/issues/6037
- https://github.com/influxdata/influxdb/issues/6041
- https://github.com/influxdata/influxdb/issues/6689
- https://github.com/influxdata/influxdb/issues/8424
- https://github.com/influxdata/influxdb/issues/8846
- https://github.com/influxdata/influxdb/issues/9045

So, if you are curious to check this, I will also be happy to learn about whether the `precision = ""` option actually works. Please note that in recent versions of InfluxDB, they might be called ns, us, ms, s, m, h already.

With kind regards,
Andreas.

amotl, 2020-04-12:
Dear Adrián,

your suggestion has been implemented by b3cb7344 and will be part of the next release.

With kind regards,
Andreas.