daq-tools/kotori #30: Custom Timestamp

GuilhermeDaniluski, 2020-05-18:
Hello, 

First of all thank you for this amazing work.

I have the need of the timestamp at the exact moment  that the data is collected, as consequence i send this timestamp along with the data value over mqtt. Is there a way to force InduxDB to record my own timestamp and not the one that it uses when it receive the data?

Thank you.

GuilhermeDaniluski, 2020-05-18:
Oh i am really sorry, i didn't read the docs enough.

You already explained what do i have to do.
https://www.getkotori.org/docs/handbook/acquisition/protocol/mqtt.html#readings-with-timestamp

closed 2020-05-18 by GuilhermeDaniluski

amotl, 2020-05-18:
Dear Guilherme,

thanks for using Kotori, we are happy it is useful to you. You might want to introspect [1] and other code around that in order to find out about the heuristics applied when ingesting and parsing the datetime field. This detail is a bit sparse on documentation yet ;].

When following the documentation at the place you've referenced [2], you will also find the section which outlines the different value types understood by Kotori [3].

Cheers,
Andreas.

[1] https://github.com/daq-tools/kotori/blob/59075812/kotori/daq/storage/influx.py#L157-L164
[2] https://www.getkotori.org/docs/handbook/acquisition/protocol/mqtt.html#readings-with-timestamp
[3] https://www.getkotori.org/docs/handbook/acquisition/timestamp.html#daq-timestamp-formats