daq-tools/kotori #33: RPi Installation

timaseth, 2020-06-04:
Hi, Lots of Home Automation enthusuasts use a Rasberry Pi (model 3, 3plus or 4) for their computing needs. I made the mistake of perhaps not reading through all steps and tried the docker install and failed.
Issue 1 was the Grafa permissions - but resolved.
Issue 2 No Mongodb for the pi
Issue 3 I read that Mongodb is option so tried 
`docker run -it --rm daqzilla/kotori kotori --version`
Nope!
```
$ docker run -it --rm daqzilla/kotori kotori --version
Unable to find image 'daqzilla/kotori:latest' locally
latest: Pulling from daqzilla/kotori
68ced04f60ab: Pull complete 
0f5503414412: Pull complete 
Digest: sha256:ff3d0a569de75fda447ad108a2ec664d8aaf545ded82ecd8c9010fc50817f94b
Status: Downloaded newer image for daqzilla/kotori:latest
standard_init_linux.go:211: exec user process caused "exec format error"
failed to resize tty, using default size
```
I am a Linux noob so maybe i am doing it all wrong or perhaps Kotori is not for the Pi.
Would love to hear from you as I am quite excited with how you have brought MQTT (even Tasmota!!) Influx and Grafana all togther.

Cheers and best wishes!

amotl, 2020-06-04:
Dear Seth,

thanks for writing in. I will have to check how we can support Kotori on ARM through Docker. In the meanwhile, would installing Kotori through a Debian package [1] feasible for you?

With kind regards,
Andreas.

[1] https://getkotori.org/docs/setup/debian.html

ryanwalexander, 2020-06-16:
I would also like to install on RaspberryPi.  I am running a stack with all the requisites which I setup with this script.  https://github.com/SensorsIot/IOTstack

Perhaps the dev would be interested in adding kotori to the script if it was ARM-ready.

amotl, 2020-06-18:
Dear Ryan,

Kotori actually is ARM-ready, see [1]. The Debian package can be installed natively or within a Docker container.

However, I see that an appropriate image should be available from DockerHub to be included within IOTstack. That would be awesome, so thanks for the heads up!

With kind regards,
Andreas.

[1] https://packages.elmyra.de/elmyra/foss/debian/pool/main/k/kotori-standard/

amotl, 2021-02-18:
Dear Seth and Ryan,

while the documentation hasn't been updated yet, I wanted to share some news here that Kotori Docker images are also available for ARM architectures now (both arm32v7 and arm64v8). I will be happy to hear about the outcome whether this works for you.

As those builds are the first of their kind, we might experience some hiccups. Because I currently don't have a RaspberryPi or similar SBC machine rigged up, I would like to humbly ask people from the community to test those images.

```sh
# Run Kotori full edition (not available for ARM unless there is demand from the community).
docker run -it --rm daqzilla/kotori kotori --version

# Run Kotori standard edition (slightly smaller images).
docker run -it --rm daqzilla/kotori-standard kotori --version
```

With kind regards,
Andreas.

amotl, 2021-07-29:
Hi again,

we just published Docker images for Kotori 0.26.10 for both arm32v7 and arm64v8 platforms. We will be happy to hear back from you about this.

### arm32v7
```shell
$ docker run -it --rm --platform=linux/arm/v7 daqzilla/kotori-standard uname -a
Linux 68c603bb0802 5.10.25-linuxkit #1 SMP Tue Mar 23 09:27:39 UTC 2021 armv7l GNU/Linux

$ docker run -it --rm --platform=linux/arm/v7 daqzilla/kotori-standard kotori --version
Kotori version 0.26.10
```

### arm64v7
```shell
$ docker run -it --rm --platform=linux/arm64/v8 daqzilla/kotori-standard uname -a
Linux ea2072063bed 5.10.25-linuxkit #1 SMP Tue Mar 23 09:27:39 UTC 2021 aarch64 GNU/Linux

$ docker run -it --rm --platform=linux/arm64/v8 daqzilla/kotori-standard kotori --version
Kotori version 0.26.10
```

With kind regards,
Andreas.

ryanwalexander, 2021-07-31:
Awesome!

closed 2022-10-15 by timaseth

amotl, 2022-10-15:
Dear Seth,

thanks for closing the issue. Does it work for you now? Let me know if you are observing any issues.

With kind regards,
Andreas.