daq-tools/kotori #34: kotori install fails on 64bit arm raspberry

hansj10, 2020-11-09:
Hi Kotori,
my pi4 running in 64bit mode, the install of kotori fails because it thinks it's on a 32bit system:

$ sudo dpkg --print-architecture
arm64

$ sudo apt-get install kotori
Reading package lists... Done
Building dependency tree
Reading state information... Done
Note, selecting 'kotori-standard:armhf' instead of 'kotori:armhf'
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 kotori-standard:armhf : Depends: python2.7:armhf but it is not going to be installed
                         Recommends: mongodb:armhf but it is not installable
                         Recommends: python-requests:armhf but it is not installable
                         Recommends: python-openssl:armhf but it is not installable
                         Recommends: python-cryptography:armhf but it is not going to be installed
                         Recommends: python-certifi:armhf but it is not installable
                         Recommends: python-numpy:armhf but it is not going to be installed
                         Recommends: python-scipy:armhf but it is not going to be installed
                         Recommends: python-matplotlib:armhf but it is not going to be installed
                         Recommends: python-tables:armhf but it is not installable
                         Recommends: python-netcdf4:armhf but it is not going to be installed
                         Recommends: libhdf5-100:armhf but it is not installable
                         Recommends: libnetcdf-c++4:armhf but it is not going to be installed
                         Recommends: libnetcdf11:armhf but it is not installable
E: Unable to correct problems, you have held broken packages.

is there no arm64 package or where in this process is armhf selected?

Of course, python is installed, I got influg/graphana setup separartly, but I'm hesitating to compile kotori...

Any hints on how to fix this?

Thanks for your help,
Hans

amotl, 2020-11-10:
Dear Hans,

thanks for writing in. You are right, we currently don't have any packages for arm64. But now that we hear that there might be some demand, we are well reconsidering this. I also want to let you know that we also started porting the code base over to Python 3.

Regarding the improvement to add arm64 to the packaging, I want to humbly point you towards [1] as well as [2]. If you are feeling lucky, you might be able to figure that out on your own behalf.

With kind regards,
Andreas.

[1] https://getkotori.org/docs/development/releasing/packaging.html
[2] https://github.com/daq-tools/kotori/search?q=armhf

amotl, 2021-02-18:
Dear Hans,

while the documentation hasn't been updated yet, I wanted to share some news here that Kotori Docker images are also available for ARM architectures now (both arm32v7 and arm64v8). I will be happy to hear about the outcome whether this works for you.

As those builds are the first of their kind, we might experience some hiccups. Because I currently don't have a RaspberryPi or similar SBC machine rigged up, I would like to humbly ask people from the community to test those images.

```sh
# Run Kotori full edition (not available for ARM unless there is demand from the community).
docker run -it --rm daqzilla/kotori kotori --version

# Run Kotori standard edition (slightly smaller images).
docker run -it --rm daqzilla/kotori-standard kotori --version
```

With kind regards,
Andreas.