daq-tools/kotori #35: "Basic example with MQTT" fails

danb35, 2020-12-08:
With apologies if I'm missing something terribly obvious (as I'm very much a n00b to MQTT, InfluxDB, and Grafana), the "basic example with MQTT" from the docs (https://getkotori.org/docs/getting-started/basic-mqtt.html) isn't working for me.

## Background
I installed Kotori on a fresh, updated Ubuntu 18 server following the instructions at https://getkotori.org/docs/setup/debian.html.  I then went to the "Getting Started" documentation and tried to follow the example with MQTT.  The snipped in `amazonas.ini` looks like this:
```
[amazonas]
enable      = true
type        = application
realm       = amazonas
mqtt_topics = amazonas/#
application = kotori.daq.application.mqttkit:mqttkit_application
```
But when I run the `mosquitto_pub` command, I get an error in the kotori log:
```
2020-12-07T19:59:39-0500 [kotori.daq.graphing.grafana.manager] INFO    : Provisioning Grafana dashboard "amazonas-ecuador" for database "amazonas_ecuador" and measurement "cuyabeno_1_sensors"
2020-12-07T19:59:39-0500 [kotori.daq.graphing.grafana.api    ] INFO    : Checking/Creating datasource "amazonas_ecuador"
2020-12-07T19:59:40-0500 [kotori.daq.services.mig            ] ERROR   : Grafana provisioning failed for storage={"node": "1", "slot": "data.json", "realm": "amazonas", "network": "ecuador", "database": "amazonas_ecuador", "measurement_events": "cuyabeno_1_events", "label": "cuyabeno_1", "measurement": "cuyabeno_1_sensors", "gateway": "cuyabeno"}, message={u'temperature': 42.84, u'humidity': 83.1}:
	[Failure instance: Traceback: <class 'grafana_api_client.GrafanaUnauthorizedError'>: Unauthorized
	/opt/kotori/lib/python2.7/site-packages/twisted/python/threadpool.py:250:inContext
	/opt/kotori/lib/python2.7/site-packages/twisted/python/threadpool.py:266:<lambda>
	/opt/kotori/lib/python2.7/site-packages/twisted/python/context.py:122:callWithContext
	/opt/kotori/lib/python2.7/site-packages/twisted/python/context.py:85:callWithContext
	--- <exception caught here> ---
	/opt/kotori/lib/python2.7/site-packages/kotori/daq/services/mig.py:269:process_message
	/opt/kotori/lib/python2.7/site-packages/kotori/daq/graphing/grafana/manager.py:129:provision
	/opt/kotori/lib/python2.7/site-packages/kotori/daq/graphing/grafana/manager.py:85:create_datasource
	/opt/kotori/lib/python2.7/site-packages/kotori/daq/graphing/grafana/api.py:104:create_datasource
	/opt/kotori/lib/python2.7/site-packages/grafana_api_client/__init__.py:73:create
	/opt/kotori/lib/python2.7/site-packages/grafana_api_client/__init__.py:64:make_request
	/opt/kotori/lib/python2.7/site-packages/grafana_api_client/__init__.py:171:make_raw_request
	]
```
I'm sure it's as a result of this that browsing to ip:3000/dashboard/db/ecuador/ fails a "Dashboard not found" message.

I'd appreciate a pointer in the right direction.

amotl, 2020-12-08:
Dear @danb35,

thanks for writing in and apologies that Kotori would not work for you without further ado. It looks like Kotori fails to authenticate on the Grafana instance when trying to provision an instant dashboard.

So, can you check that the configuration in `/etc/kotori/kotori.ini` matches the credentials of your Grafana?

https://github.com/daq-tools/kotori/blob/b3cb7344a37f806d25f5bcd844ff0052b07bd01a/etc/production.ini#L39-L43

Let me know if you need further help on this.

With kind regards,
Andreas.

danb35, 2020-12-08:
Thanks for the response--as I'd expected, it turned out to be something quite simple.  I'd already logged in to Grafana, and when I did so, it required that I change the password for `admin`.  Once I updated `kotori.ini` to match that, the test data was saved:
```
2020-12-08T04:54:29-0500 [kotori.daq.services.mig            ] INFO    : [amazonas    ] transactions: 0.00 tps
2020-12-08T04:54:54-0500 [kotori.daq.graphing.grafana.manager] INFO    : Provisioning Grafana dashboard "amazonas-ecuador" for database "amazonas_ecuador" and measurement "cuyabeno_1_sensors"
2020-12-08T04:54:54-0500 [kotori.daq.graphing.grafana.api    ] INFO    : Checking/Creating datasource "amazonas_ecuador"
2020-12-08T04:54:54-0500 [kotori.daq.graphing.grafana.api    ] INFO    : response: {u'message': u'Datasource added', u'datasource': {u'readOnly': False, u'name': u'amazonas_ecuador', u'database': u'amazonas_ecuador', u'url': u'http://localhost:8086/', u'basicAuth': False, u'jsonData': {}, u'access': u'proxy', u'isDefault': False, u'typeLogoUrl': u'', u'orgId': 1, u'user': u'root', u'version': 1, u'basicAuthUser': u'', u'secureJsonFields': {}, u'password': u'root', u'type': u'influxdb', u'id': 1, u'basicAuthPassword': u'', u'withCredentials': False}, u'name': u'amazonas_ecuador', u'id': 1}
2020-12-08T04:54:54-0500 [kotori.daq.graphing.grafana.api    ] INFO    : Get folder with uid="instagraf"
2020-12-08T04:54:54-0500 [kotori.daq.graphing.grafana.api    ] INFO    : Getting dashboard "amazonas-ecuador"
2020-12-08T04:54:54-0500 [kotori.daq.graphing.grafana.api    ] INFO    : Client Error 404: {"message":"Dashboard not found"}
2020-12-08T04:54:54-0500 [kotori.daq.graphing.grafana.api    ] INFO    : Creating/updating dashboard "amazonas-ecuador"
2020-12-08T04:54:55-0500 [kotori.daq.graphing.grafana.api    ] INFO    : Grafana response: {u'status': u'success', u'uid': u'hpIYBZJGz', u'url': u'/d/hpIYBZJGz/amazonas-ecuador', u'slug': u'amazonas-ecuador', u'version': 1, u'id': 2}
2020-12-08T04:54:55-0500 [kotori.daq.graphing.grafana.api    ] INFO    : Checking dashboard "amazonas-ecuador"
```
I'm able to see the instant dashboard by navigating through the Grafana pages (from the home page, manage dashboards, # Instant Dashboards, amazonas-ecuador), but the URL mentioned in the docs of ip:3000/dashboard/db/ecuador/ still gives the same "Dashboard not found" message.  It looks like that URL needs to be updated to ip:3000/dashboard/db/amazonas-ecuador/

amotl, 2020-12-08:
Thanks for reporting this works for you now. I've also updated the documentation with 4bf1cfe6b7.

closed 2020-12-10 by danb35