daq-tools/kotori #36: Add documentation - Caddy reverse proxy

danb35, 2020-12-10:
I see that your docs include a page for setup behind a Nginx reverse proxy (https://getkotori.org/docs/setup/nginx.html), but I think it'd be nice to include instructions for using Caddy as well.  I use it in a number of applications, and the big benefits I see are (1) it automatically manages TLS, including obtaining and renewing certificates from Let's Encrypt, and implementing a sensible and secure TLS configuration (the defaults give an A rating using https://github.com/drwetter/testssl.sh); and (2) much shorter and simpler configuration files (the complete server configuration for this application, including all the TLS stuff, can be as short as 10 lines).

The one downside is that in many cases it needs to be built from source--though as it's written in Go, that isn't as big of an issue as it could be.

I've written up a draft of a guide--feel free to adopt it wholesale, modify as appropriate, link to it, or whatever:
https://www.familybrown.org/dokuwiki/doku.php?id=advanced:kotori_caddy

amotl, 2020-12-10:
Hi again,

thanks a stack for this contribution. I will be happy to link to it from Kotori's documentation or might even include the snippets there.

Are you aware that Kotori also optionally sports a HTTP API, defaulting to localhost:24642? It would be nice to also include that into those snippets.

With kind regards,
Andreas.

danb35, 2020-12-10:
> Are you aware that Kotori also optionally sports a HTTP API, defaulting to localhost:24642? It would be nice to also include that into those snippets.

I hadn't run across that yet, but from a quick look at the docs, it looks like all calls to the API would begin with `/api/`, right?  In that case, it should be very easy to add--you'd just have two `reverse_proxy` lines, like this:
```
        reverse_proxy /api/* localhost:24642
        reverse_proxy localhost:3000
```
This should (untested as yet) send any requests for /api/foo to port 24642, and everything else to port 3000.

amotl, 2020-12-11:
Thanks! I will give that a short testdrive when dedicating this a respective section within the documentation.