daq-tools/kotori #37: Kotori server does not start

Cr4ck3r32, 2021-01-16:
Good morning, I am new to the installation part of Kotori, I have followed all the steps on your page but I get to the moment where I have to enter the server with http://kotori.example.org:3000/ and it does not work. I've started the server with "systemctl start kotori" and it doesn't work either. When I get to the step of the tutorial where you have to put the command: mosquitto_pub -t $ CHANNEL_TOPIC -m '{"temperature": 42.84, "humidity": 83.1}' in the terminal I get this: "Error: Connection Refused"

Can someone help me with this please?

Thank you

amotl, 2021-01-17:
Hi @Cr4ck3r32,

thanks for writing in.

First of all, let me tell you a bit how Kotori bundles different best-of-breed FOSS software around itself.

On port 3000, there should be a Grafana instance listening. While we used `kotori.example.org` within the documentation, you should try `localhost:3000` on your workstation.

Then, when trying to publish a telemetry message, you need an MQTT broker listening on `localhost:1883`. We chose `mosquitto`, but in theory you could use any other. Can you verify this is the case?

Did you actually install the Debian package(s) by following the documentation at [1] on your operating system? If so, we might have to check whether Kotori itself is actually running.

To check if your system actually runs the respective processes Mosquitto, Grafana and Kotori, you might want to type something along the lines of `sudo ps axf` and check the output.

With kind regards,
Andreas.

[1] https://getkotori.org/docs/setup/debian.html

Cr4ck3r32, 2021-01-17:
Hi Andreas,

After trying "localhost: 3000" grafana has started. But I understand that now I have to tell you where to get the data from to show it, right? In this case I have to go to plugins and select Influxdb and fill in the parameters. But when filling in the parameters in the Influxdb window below in "database", "user" and "password" what is the data here, because I have not seen it specified anywhere. Or is this not the process and am I getting wrong?

After testing the services with the command "ps axf | egrep -i '(mosquitto | grafana | kotori)'" what it shows me in the terminal is:

root@kotori:/home/kotori# ps axf | egrep -i '(mosquitto|grafana|kotori)'
    475 ?        Ss     0:00 avahi-daemon: running [kotori.local]
    669 ?        Ssl    0:02 /usr/sbin/grafana-server --config=/etc/grafana/grafana.ini --pidfile=/var/run/grafana/grafana-server.pid --packaging=deb cfg:default.paths.logs=/var/log/grafana cfg:default.paths.data=/var/lib/grafana cfg:default.paths.plugins=/var/lib/grafana/plugins cfg:default.paths.provisioning=/etc/grafana/provisioning
   3010 pts/0    S+     0:00                      \_ grep -E --color=auto -i (mosquitto|grafana|kotori)
   1883 ?        Ss     0:00 /bin/sh -c /opt/kotori/bin/kotori --config /etc/kotori/kotori.ini  >>/var/log/kotori/kotori.log 2>>/var/log/kotori/kotori.log
   1886 ?        S      0:00  \_ /opt/kotori/bin/python /opt/kotori/bin/kotori --config /etc/kotori/kotori.ini

I understand that all 3 services work, right? But i still get this error when i put in the terminal this command:

mosquitto_pub -t $CHANNEL_TOPIC -m '{"temperature": 42.84, "humidity": 83.1}'

in the terminal I get this: "Error: Connection Refused"

Can you explain to me what am i doing wrong ? Maybe I don't understand how the 3 services work together.

Thank you very much and have a nice day.

amotl, 2021-01-17:
Hi,

we will make it! Let me give you more information based on what you are reporting.

### Grafana
> After trying `"localhost: 3000"`, [I believe that] Grafana has started.

Great. Please be aware that after logging in using `admin/admin`, you are asked to change the password. If you are actually doing so (you can also skip this step), you will have to adjust the credentials within `/etc/kotori/kotori.ini` and restart the service using `systemctl restart kotori`.

> But I understand that now I have to tell you where to get the data from to show it, right?

One of the main features of Kotori is that it will provision Grafana automatically on the arrival of data on the MQTT bus in order to give users an "instant-on" experience. That's why it needs the appropriate credentials in order to log into Grafana on your behalf.

### Mosquitto
Based on the output of the process lists you are sharing, I can't see `mosquitto` is running. That's actually why you receive  `Error: Connection Refused` when invoking `mosquitto_pub`.

Maybe you will just have to give it a start signal like `systemctl start mosquitto`. While I documented that fact at [1] for InfluxDB and Grafana, it might also be needed for Mosquitto. Please let me know if that helps already.

With kind regards,
Andreas.

[1] https://getkotori.org/docs/setup/debian.html#setup-the-whole-software-stack

amotl, 2021-01-19:
Hi @Cr4ck3r32,

have you been able to make any more progress? Let me know if you need further assistance.

With kind regards,
Andreas.

amotl, 2021-07-29:
Dear @Cr4ck3r32,

`systemctl start mosquitto` should start Mosquitto on your machine, I believe this is the missing link. I am closing this issue now, please feel free to reopen if you have further questions.

With kind regards,
Andreas.

closed 2021-07-29 by amotl