daq-tools/kotori #45: Docker image 0.25.0 /  grafana 7.4 not compatible with 0.24.5

scube38, 2021-02-08:
Hello, 

Just tried to upgrade to grafana 7.4 with latest kotori image 0.24.5   but  GraphanaClient APi not compatible and reporting error.  I've seen a 0.25.0 been tagged with grafana 7.4 support,  any chance to make a try with an updated  kotori docker image updated to 25.0 ? 
Thanks 
Best Regards, PLo
 =>> BTW, great piece of software !!

amotl, 2021-02-08:
Hi PLo,

thanks for writing in.

Ha! You just caught us in the act of giving Kotori some love again by a) porting it finally to Python 3 (long overdue) and b) also making it compatible with recent versions of the upstream infrastructure components (InfluxDB 2.x and Mosquitto 2.x still pending).

We will see what we can do to push new Docker images for Kotori version 0.25.0 soon.

With kind regards,
Andreas.

P.S.:

> BTW, great piece of software !!

Thank you so much, very much appreciated. While being at it, don't forget to add a star on GitHub ;] and spread the word. At the same time, I would like to humbly ask if you can share one or more use cases where you are actually using Kotori? In this manner, we can improve the gallery within the documentation [1] in a similar spirit like what @DmitrySidorow and @jm66 added to https://github.com/panodata/grafana-map-panel/issues/90 just recently.

[1] https://getkotori.org/docs/gallery/

amotl, 2021-02-10:
Hi PLo,

`daqzilla/kotori:0.26.0-preview` was just uploaded. I haven't been able to test it, but at least it seems to respond slightly.
```sh
$ docker run -it --rm daqzilla/kotori:0.26.0-preview bash

root@c9d9ef90e3b1:/# kotori --version
Kotori version 0.26.0
```

Please let me know when all is working well on your end.

With kind regards,
Andreas.

scube38, 2021-02-12:
Thnaks a lot Andreas, will give it a test . Appreciate your prompt help.  Thx , Ploq.

amotl, 2021-02-13:
Hi Ploq,

@wetterfrosch also just asked me what it would take to upgrade our software stack behind [1,2] to run Grafana 7.4. So, we will be very happy to hear about the outcome on your end.

In the meanwhile, I will go back to work on the packaging infrastructure in order to also provide new distribution packages (`.deb` files) for Debian and Ubuntu systems. To streamline that, I've started to improve the package building on behalf of [3] with f0b37d3.

With kind regards,
Andreas.

cc @timaseth, @ryanwalexander (#33) and @hansj10 (#34)

[1] https://swarm.hiveeyes.org/
[2] https://weather.hiveeyes.org/
[3] https://github.com/daq-tools/kotori/blob/master/fabfile.py

amotl, 2021-02-18:
Hi Ploq,

I've just retracted the `0.26.0-preview` tag and published a regular release `0.26.6` along the lines with corresponding `kotori-standard` images for `linux/arm/v7` and `linux/arm64/v8` platforms and also tagged them with `latest`.

While the documentation hasn't been updated yet, I wanted to share the news here and will be happy to hear about the outcome whether this works for you. As those builds are the first of their kind, we might experience some hiccups. Because I currently don't have a RaspberryPi or similar SBC machine rigged up, I would like to humbly ask people from the community to test those images.

```sh
# Run Kotori full edition (not available for ARM unless there is demand from the community).
docker run -it --rm daqzilla/kotori kotori --version

# Run Kotori standard edition (slightly smaller images).
docker run -it --rm daqzilla/kotori-standard kotori --version
```

With kind regards,
Andreas.

amotl, 2021-07-29:
Dear Ploq,

we just released Kotori 0.26.10 as `daqzilla/kotori` as well as `daqzilla/kotori-standard`. The releases are compatible with Grafana 7. Support for Grafana 8 is also in the making, see #57.

With kind regards,
Andreas.