daq-tools/kotori #49: Connection refused via MQTT

danb35, 2021-03-30:
I've been using Kotori for a few months to log data from my WeeWx system via MQTT, and it's been working well--until yesterday afternoon.  I rebooted the system running Kotori at that time, and weewx hasn't logged any records since that.  Sadly, it also doesn't log any error messages.

However, I'm noticing that I get "connection refused" messages for remote MQTT requests:
```
(mppsolar) root@solar:~# mosquitto_pub -t 'GS/topic' -m 'helloWorld' -h kotori
Error: Connection refused
```
Mosquitto appears to be running on the kotori system:
```
root@kotori:/var/log# systemctl status mosquitto
● mosquitto.service - Mosquitto MQTT Broker
   Loaded: loaded (/lib/systemd/system/mosquitto.service; enabled; vendor preset: enabled)
   Active: active (running) since Tue 2021-03-30 10:45:34 EDT; 1h 36min ago
     Docs: man:mosquitto.conf(5)
           man:mosquitto(8)
  Process: 677 ExecStartPre=/bin/chown mosquitto: /var/run/mosquitto (code=exited, status=0/SUCCESS)
  Process: 659 ExecStartPre=/bin/mkdir -m 740 -p /var/run/mosquitto (code=exited, status=0/SUCCESS)
  Process: 657 ExecStartPre=/bin/chown mosquitto: /var/log/mosquitto (code=exited, status=0/SUCCESS)
  Process: 609 ExecStartPre=/bin/mkdir -m 740 -p /var/log/mosquitto (code=exited, status=0/SUCCESS)
 Main PID: 678 (mosquitto)
    Tasks: 1 (limit: 2316)
   CGroup: /system.slice/mosquitto.service
           └─678 /usr/sbin/mosquitto -c /etc/mosquitto/mosquitto.conf

Mar 30 10:45:34 kotori systemd[1]: Starting Mosquitto MQTT Broker...
Mar 30 10:45:34 kotori systemd[1]: Started Mosquitto MQTT Broker.
root@kotori:/var/log# 
```
And it appears to be listening on port 1883:
```
root@kotori:/var/log# netstat -tulpn
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       PID/Program name    
tcp        0      0 127.0.0.1:1883          0.0.0.0:*               LISTEN      678/mosquitto       
tcp        0      0 127.0.0.1:24642         0.0.0.0:*               LISTEN      1312/python         
tcp        0      0 127.0.0.1:2019          0.0.0.0:*               LISTEN      671/caddy           
tcp        0      0 127.0.0.1:27017         0.0.0.0:*               LISTEN      596/mongod          
tcp        0      0 127.0.0.53:53           0.0.0.0:*               LISTEN      500/systemd-resolve 
tcp        0      0 0.0.0.0:22              0.0.0.0:*               LISTEN      718/sshd            
tcp        0      0 127.0.0.1:8088          0.0.0.0:*               LISTEN      672/influxd         
tcp6       0      0 :::443                  :::*                    LISTEN      671/caddy           
tcp6       0      0 ::1:1883                :::*                    LISTEN      678/mosquitto       
tcp6       0      0 :::80                   :::*                    LISTEN      671/caddy           
tcp6       0      0 :::8086                 :::*                    LISTEN      672/influxd         
tcp6       0      0 :::22                   :::*                    LISTEN      718/sshd            
tcp6       0      0 :::3000                 :::*                    LISTEN      662/grafana-server  
udp        0      0 127.0.0.53:53           0.0.0.0:*                           500/systemd-resolve 
udp        0      0 192.168.1.68:68         0.0.0.0:*                           445/systemd-network 
```

Nothing really stands out in the kotori log, though its size is almost 4 GB, so I could easily have missed it.

amotl, 2021-03-30:
Dear @danb35,

thank you for your report. I believe you might got tripped by the same thing we experienced recently. Mosquitto has been upgraded to version 2.x, which brings in a more secure default configuration. You will explicitly have to configure it like outlined at [1].

Please let me know if this resolves your problem already or whether we will have to dig deeper why the data acquisition stopped working.

With kind regards,
Andreas.

[1] https://community.hiveeyes.org/t/giving-the-backend-software-infrastructure-some-love/3839/5

---

### Mosquitto 2.0

Please note that Mosquitto 2.x introduced a breaking change. However, it was a good decision because it strengthens the security on default installations.

> When the Mosquitto broker is run without configuring any listeners it will now bind to the loopback interfaces `127.0.0.1` and/or `::1`. This means that only connections from the local host will be possible.

So, in order to make Mosquitto listen on the public interface and accept anonymous requests again, you will have to configure your `/etc/mosquitto/conf.d/default.conf` like that:

https://github.com/eclipse/mosquitto/blob/12ff9d5e/docker/2.0/mosquitto-no-auth.conf

```conf
# This is a Mosquitto configuration file that creates a listener on port 1883
# that allows unauthenticated access.

listener 1883
allow_anonymous true
```

#### References
- [ Version 2.0.0 released » Breaking changes](https://mosquitto.org/blog/2020/12/version-2-0-0-released/#breaking-changes)
- [Migrating from 1.x to 2.0 » Listener behaviour changes](https://mosquitto.org/documentation/migrating-to-2-0/#listener-behaviour-changes)

danb35, 2021-03-31:
That was it indeed.  Now to figure out how to tell [mpp-solar](https://github.com/jblance/mpp-solar) how to send data in a format Kotori will read...

closed 2021-03-31 by danb35