daq-tools/kotori #52: Weewx

18Scoobz, 2021-06-02:
I have everything set up per these instructions.
https://getkotori.org/docs/examples/weewx.html

The exception is that I am using a mqtt broker on another server.

Kotori logs did not show any issue with the broker

now all I have is these logs I am assuming they are the mqtt messages

2021-06-02T23:33:32+0000 [kotori.daq.services.mig            ] INFO    : [weewx       ] transactions: 0.00 tps
2021-06-02T23:34:32+0000 [kotori.daq.services.mig            ] INFO    : [weewx       ] transactions: 0.00 tps
2021-06-02T23:35:32+0000 [kotori.daq.services.mig            ] INFO    : [weewx       ] transactions: 0.00 tps
2021-06-02T23:36:32+0000 [kotori.daq.services.mig            ] INFO    : [weewx       ] transactions: 0.00 tps

What do I do to get information into the database?  Then Grafana?

18Scoobz, 2021-06-07:
figured it out

closed 2021-06-07 by 18Scoobz

amotl, 2021-06-25:
Hi @18Scoobz,

thank you for writing in and congratulations that you figured it out.

With kind regards,
Andreas.