daq-tools/kotori #53: How to pull mqtt data from 2 topics/weather stations

18Scoobz, 2021-06-10:
I am trying to pull mqtt data from 2 topics.  What is the format in weewx.ini.  I have 2 weather stations I want to compare.

amotl, 2021-06-25:
Hi @18Scoobz,

when you look at the configuration snippet of WeeWx's `weewx.conf` [1], you will just have to add a different topic, like
```
topic = weewx/097287c4-6fb0-4aeb-a095-00d65ecb15f7/Leoni/VantagePro2
```

As you can see in Kotori's `weewx.ini`, the prefix `weewx` is fixed. Everything which comes after this prefix can be freely set by the user of a data channel (you!). However, you should adhere to the current topology and use three "path" components after the first-level prefix / realm.

On everything else you have absolute freedom and you can acquire data for any number of devices with the same Kotori configuration. In other words: You don't have to individually add each and every channel to Kotori's configuration.

I dearly hope it will work out and this guideline will be helpful for you.

With kind regards,
Andreas.

[1] https://getkotori.org/docs/gallery/weewx.html#setup-weewx-mqtt-plugin

amotl, 2021-07-11:
Hi @18Scoobz,

did you have success submitting measurement data from two or more instances of WeeWX to Kotori? Let me know if you need further assistance.

With kind regards,
Andreas.