daq-tools/kotori #69: Processing MQTT message failed: 'NoneType' object has no attribute 'endswith'

amotl, 2021-11-20:
Hi,

people using Kotori reported this error message to us:
```log
2021-11-20T20:25:51.019107+0100 [kotori.daq.services.mig            ] DEBUG   : Processing message on topic 'workbench/testdrive/area-42/evb-ea-ind-02' with payload '{"bme280_temp":17.9,"bme280_humidity":39.80176,"bme280_pressure":938.2483}'
2021-11-20T20:25:51.019295+0100 [kotori.daq.services.mig            ] DEBUG   : Topology address: {'realm': 'workbench', 'network': 'testdrive', 'gateway': 'area-42', 'node': 'evb-ea-ind-02', 'slot': None}
2021-11-20T20:25:51.019605+0100 [kotori.daq.services.mig            ] ERROR   : Processing MQTT message failed from topic "workbench/testdrive/area-42/evb-ea-ind-02":
        [Failure instance: Traceback: <class 'AttributeError'>: 'NoneType' object has no attribute 'endswith'
```

```python
[Failure instance: Traceback: <class 'AttributeError'>: 'NoneType' object has no attribute 'endswith'
    /usr/lib/python3.9/threading.py:954:_bootstrap_inner
    /usr/lib/python3.9/threading.py:892:run
    /home/tonke/development/kotori/.venv/lib/python3.9/site-packages/twisted/_threads/_threadworker.py:46:work
    /home/tonke/development/kotori/.venv/lib/python3.9/site-packages/twisted/_threads/_team.py:190:doWork
    --- <exception caught here> ---
    /home/tonke/development/kotori/.venv/lib/python3.9/site-packages/twisted/python/threadpool.py:250:inContext
    /home/tonke/development/kotori/.venv/lib/python3.9/site-packages/twisted/python/threadpool.py:266:<lambda>
    /home/tonke/development/kotori/.venv/lib/python3.9/site-packages/twisted/python/context.py:122:callWithContext
    /home/tonke/development/kotori/.venv/lib/python3.9/site-packages/twisted/python/context.py:85:callWithContext
    /home/tonke/development/kotori/kotori/daq/services/mig.py:135:process_message
    /home/tonke/development/kotori/kotori/daq/services/mig.py:185:decode_message
    /home/tonke/development/kotori/kotori/daq/decoder/__init__.py:27:probe
```

With kind regards,
Andreas.

amotl, 2021-11-20:
Thanks for the report! It looks like the topic suffix `/data.json` would be missing? We will keep this as a note to eventually improve the error message.

amotl, 2021-11-20:
After adding the `/data.json` suffix, the data acquisition process works flawlessly.

```log
2021-11-20T20:30:45.112023+0100 [kotori.daq.services.mig            ] DEBUG   : Processing message on topic 'workbench/testdrive/area-42/evb-ea-ind-02/data.json' with payload '{"bme280_temp":17.95,"bme280_humidity":39.52832,"bme280_pressure":938.1571,"time":1637436645}'
2021-11-20T20:30:45.112247+0100 [kotori.daq.services.mig            ] DEBUG   : Topology address: {'realm': 'workbench', 'network': 'testdrive', 'gateway': 'area-42', 'node': 'evb-ea-ind-02', 'slot': 'data.json'}
2021-11-20T20:30:45.112457+0100 [kotori.daq.services.mig            ] DEBUG   : Storage location: {'realm': 'workbench', 'network': 'testdrive', 'gateway': 'area-42', 'node': 'evb-ea-ind-02', 'slot': 'data.json', 'label': 'area_42_evb_ea_ind_02', 'database': 'workbench_testdrive', 'measurement': 'area_42_evb_ea_ind_02_sensors', 'measurement_events': 'area_42_evb_ea_ind_02_events'}
2021-11-20T20:30:45.146935+0100 [kotori.daq.storage.influx          ] DEBUG   : Storage success: {'measurement': 'area_42_evb_ea_ind_02_sensors', 'tags': {}, 'time_precision': 's', 'time': 1637436645, 'fields': {'bme280_temp': 17.95, 'bme280_humidity': 39.52832, 'bme280_pressure': 938.1571}}
```

wetterfrosch, 2023-12-09:
I am stumbling upon devices from the manufacturer shelly where the last part of the realm is always expanded with `/info` or `/settings,` like `realm/of/topic/data.json/info`. Is for those devices a tweak in the config file possible?

amotl, 2024-01-21:
Hi @wetterfrosch,

I responded to your report over at GH-163, where we are looking at unlocking support for Shelly devices.

With kind regards,
Andreas.