daq-tools/kotori #92: InfluxDB 2.x ?

nadolny, 2022-07-11:
Kotori is a great tool. Is an update that supports InfluxDB 2.x in your future plans?

nadolny, 2022-07-11:
Dear Andreas,

Thank you for your quick response.
At the moment evaluating Kotori to be used in an IoT solution for use in Industry.
In the scenario of this evaluation, only the first step would be excellent

1.Receive/ingest data payloads and write them to InfluxDB 2.x buckets.

Let me know if in addition to the points you mentioned, there would also be an update on the project

https://github.com/daq-tools/pyinfluxql  => Flux

With kind regards,
Aurelius

amotl, 2022-07-11:
Hi again,

> In addition to the points you mentioned, [there would also be the [export feature], which would need to be prepared for InfluxDB 2.x, no?].

You are absolutely right. [^4] and [^5] are two other spots which would need to be adjusted. Thank you for mentioning that!

[^4]: https://github.com/daq-tools/kotori/blob/main/kotori/io/protocol/influx.py
[^5]: https://github.com/daq-tools/kotori/blob/main/kotori/io/export/influx.py

> Would we need to make the [PyInfluxQL] package understand Flux?

I am not sure about it. Maybe the semantics don't fit well together, as InfluxDB 2.x has different addressing (database vs. bucket) _and_ different query language structure (InfluxQL vs. Flux). Maybe it would become a whole mess when trying to squeeze all into the same package. We would have to evaluate this detail when spending some more thoughts on this topic.

With kind regards,
Andreas.

[export feature]: https://getkotori.org/docs/handbook/export/
[PyInfluxQL]: https://github.com/daq-tools/pyinfluxql

nadolny, 2022-07-11:
Hi again,

We could update the "influxdb-python client library" module by the "influxdb-client-python" module

In the link https://github.com/influxdata/influxdb-client-python We check a note:

Note: Use this client library (influxdb-client-python) with InfluxDB 2.x and InfluxDB 1.8+.

The fact that kotori is using influxdb 1.8 wouldn't have some facilities?

With kind regards,
Aurelius

amotl, 2022-07-12:
Hi Aurelius,

thank you for sharing this detail. Yes, using that client library was planned for supporting InfluxDB 2.x. The section at [1] has some more details about compatibility with InfluxDB 1.8, where it is stated that it will also use the new API based on the "buckets" addressing scheme.

Currently, Kotori supports InfluxDB 1.6 - 1.8, through the previous "database/collection" addressing scheme. I am not objecting against dropping support for the InfluxDB 1.x series, but I would favor to have them both supported in parallel. Actually, we are still running InfluxDB 1.7 in production ;].

With kind regards,
Andreas.

[1] https://github.com/influxdata/influxdb-client-python#influxdb-1-8-api-compatibility

amotl, 2022-07-18:
Hi again,

I just want to share some additional thoughts on this topic. I believe, when starting to work on this feature/improvement, we should aim for converging all that InfluxDB-specific code scattered throughout the code base into a subsystem with a more well-defined interface.

Based on that interface, data source adapters for other databases could be implemented more easily. For example, I would like to add support for CrateDB [^1] or other PostgreSQL-compatible databases in the future as well.

With kind regards,
Andreas.

[^1]: https://github.com/crate/crate