daq-tools/kotori #104: docker-compose up is taking very long

BabylonCoder, 2022-12-03:
Hi, anyone know why the docker-compose up is taking ages to unpack the Mosquitto, InfluxDB, MongoDB and Grafana ? is that normal ? I have MacOs BigSur ! thank you

BabylonCoder, 2022-12-24:
Sorted, this was my installation method

closed 2022-12-24 by BabylonCoder

amotl, 2022-12-24:
Hi @BabylonCoder,

apologies for the late reply. Did you manage to get the stack running properly in the end? Please let me know if you need further assistance, and if you find any bugs or spots for improvement.

With kind regards,
Andreas.

BabylonCoder, 2022-12-25:
Hi @amotl; all good. It was my installation. It would be great, though,  to do a detailed; video so we can learn more and also so I can teach it to my class at university. Thank you for a great product.