daq-tools/kotori #130: Docker is sunsetting Free Team organizations

amotl, 2023-03-14:
I've just received this email from Docker Inc.

> #### Docker is sunsetting Free Team organizations
>
> Free Team organizations are a legacy subscription tier that no  longer exists. This tier included many of the same features, rates, and  functionality as a paid Docker Team subscription.
>
> After reviewing the list of accounts that are members of legacy  Free Team organizations, we’ve identified yours as potentially being one  of them.
>
> If you own a legacy Free Team organization, access to paid features — including private repositories — will be suspended on **April 14, 2023 (11:59 pm UTC)**. Upgrade your subscription before  **April 14, 2023** to continue accessing your organization.
>
> If you don’t upgrade to a paid subscription, Docker will retain  your organization data for 30 days, after which it will be subject to  deletion. During that time, you will maintain access to any images in  your public repositories, though rate limitations  will apply. At any point during the 30-day period, you can restore  access to your organization account if you upgrade to a paid  subscription. Visit our [FAQ](https://web.docker.com/rs/790-SSB-375/images/privatereposfaq.pdf) for more information.

So, this probably will make https://hub.docker.com/r/daqzilla/kotori cease to exist. We will look into publishing the OCI images on the [GitHub Container Registry (GHCR)](https://docs.github.com/en/packages/working-with-a-github-packages-registry/working-with-the-container-registry) instead.

Vad1mo, 2023-03-15:
If, for some reason, the Docker-Sponsored Open-Source Program isn't working out, we are happy to help you out. We operate a Harbor-based [container registry as a service ](https://container-registry.com/)that has many benefits over most of the other registries out there.

There are also features regarding containerized image distribution, that might be valuable too as well.

amotl, 2023-03-16:
Dear Vadim,

this sounds interesting. We are maintaining a few open source projects which publish OCI images to GHCR listed below, and a few others will probably join those. While we are currently happy with publishing the images to GHCR, we will also be curious to try out C8N.

Do we understand correctly that you offer free organizational accounts for open source projects, which Docker Inc. also did, before announcing the sunset?

With kind regards,
Andreas.

P.S.: For staging OCI images through GHA to C8N, it would be just a matter of adjusting this section within the corresponding build recipes, right?

#### Publishing to different registry via GHA
```yaml
- name: Login to GHCR
  uses: docker/login-action@v2
  with:
    registry: ghcr.io
    username: ${{ github.repository_owner }}
    password: ${{ github.token }}
```
-- [docker-publish-standard.yml#L122-L127](https://github.com/earthobservations/wetterdienst/blob/v0.54.1/.github/workflows/docker-publish-standard.yml#L122-L127)

#### OCI images on GHCR
- https://github.com/daq-tools/skeem
- https://github.com/earthobservations/wetterdienst
- https://github.com/jpmens/mqttwarn
- https://github.com/panodata/grafana-wtf

amotl, 2023-03-26:
On Friday, I've received this message.

> #### We’re no longer sunsetting Free Team organizations
> On March 14, 2023, we emailed you about your Free Team subscription, outlining our intention to sunset that plan. After listening to the concerns of the community, we’ve decided to reverse course, and are no longer sunsetting the Free Team plan.
>
> If you’re currently on the Free Team plan, you no longer have to migrate to another plan by April 14. All customers who upgraded to a paid subscription will automatically receive a full refund for the transaction in the next 30 days, allowing them to use their new paid subscription for free for the duration of the term they purchased.
>
> We apologize for the alarm our decision caused. For more details, please [visit our FAQ](https://www.docker.com/developers/free-team-faq/).

amotl, 2023-03-26:
However, apparently there are a few restrictions for the "Docker Free Team plan":

> #### What’s the difference between the (paid) Docker Team plan and the Docker Free Team plan?
>
> The main advantages of (paid) Docker Team vs. Docker Free Team are:
>
> - Unlimited number of teams (vs. 1 team in Free Team)
> - Up to 100 seats total (vs. 3 seats max in Free Team)
> - Unlimited private repositories (vs. no private repositories included in Free Team)
> - 5,000 pulls/day/user (vs. 200 pulls/6 hours/user in Free Team)
> - The Free Team plan does not [include a license for commercial use of Docker Desktop](https://www.docker.com/pricing/faq/#subscriptionandlicensing) at a company of more than 250 employees OR more than $10 million in annual revenue.

_200 pulls/6 hours/user_ is probably enough for regular use, but what happens when there are peak times in demand?

closed 2024-04-06 by amotl